QIDITECH/QIDISlicer #101: Updated to 1.1.7...files won't upload?

Phoeff99, 2024-11-10:
Updated as stated and firmware and configuration files are up to date. When I click "Upload and Print" I get an immediate message COMPLETED regardless of file size and noting is uploaded. Sometime is get "filename --> 192.168.1.x.10088.  Still have to wait 0:00:00" and no file is uploaded. Selected correct physical printer with correct IP....any help? resorting to USB transfer in the meantime, but wifi connectivity would be preferred.

QIDITECH, 2024-11-11:
Does it work in older versions?
This seems to be a network problem, please try another router.
Please tell me the system version and configuration of your computer.

Phoeff99, 2024-11-11:
Yes it worked fine with ver 1.1.5 (I never upgraded to 1.1.6) and only maybe once has it worked since I upgraded to 1.1.7 and that was after a long wait. The connectivity with moonraker (if that's the software which allows me to monitor the printer function, temperature video stream etc) is very sporadic and only connects usually for a few seconds at a time before trying to reconnect. Once the print is complete the connection remains stable but it still does not allow me to upload another print file. The connectivity was not as sporadic previously with 1.1.5.

As for router, you mean actual router device (hardware) or another IP address assigned to the printer? I don't have additional routers to try i'm afraid. Asnd please excuse my ignorance with these things, but for computer system version and configuration, I am running MAC OS Sonoma 14.5. Is that what you mean? I am not sure what other details you need but happy to provide them if you can be more specific. Again, i'm not a computer engineer type, sorry if that's not what you meant. 

I have ordered USB wifi adapter which I understand from other users on various QIDI discussion groups has helped wifi connectivity. 

If it is a router or signal issue i'm hoping that adapter will resolve it, but I am not sure why the upgrade to 1.1.7 would have caused it as neither the printer nor router have changed location. Appreciate your help and patience.

QIDITECH, 2024-11-12:
Thanks for your cooperation, we retested this part of the code on macos, but did not find any problems. We are not sure which link (software, network environment, hardware) is at fault, so we need your cooperation to determine the problem.
1. Please reinstall 1.1.5 and test it. If 1.1.5 still works, it indicates that 1.1.7 is the problem.
2. Click the Device TAB to go to the fluidd page and check whether the fluidd page of the two software versions is connected properly.
3. Replace your wifi adapter (which you already bought).

QIDITECH, 2024-11-13:
Q1 Pro supports QIDI Link mobile APP connection, after connection can access the Internet (please update the firmware to the latest version). Physical printers are connected using a local area network. LAN connections are more stable and fast, but they cannot be connected across networks.
If you select only one printer, you do not need to set the number of simultaneous sends and the interval.