QIDITECH/QIDISlicer #109: Slicer 1.2.0 does not load .step file

cptcl, 2024-11-29:
Hi there.
Today i downloaded and installed 1.2.0
Since then i can open .stl and .3mf files as usual

But if i double click an .step file, qidislicer wont open.. WIndows just loads and nothing more happen.
if i open qidislicer manualy and try to import a .step file i get this

[image]

and it does not move. Qidi slicer cannot be closed except over taskmanager
 
Reverting back to 1.1.7 and now i can open .step files again
What can i do to use 1.2.0?

Win 11 Pro 23H2

QIDITECH, 2024-11-30:
1.2.0 as a test version, there are still some bugs that have not been fixed, we will release the next version after repair.

wjyLearn, 2024-12-20:
The bug has been fixed in the latest version.