QIDITECH/QIDISlicer #111: ver. 1.2.0 on MacOS BigSur 11.7.10 (x86) crashes when sending to printer

mssbg, 2024-12-08:
I have X-Plus 3, updated to latest firmware 4.2.15. Updated the slicer to latest ver. 1.2.0 
Open a .step file, slice it as usual. 
Click export and send to printer
The dialog to send it to printer shows up, I select my only printer and click send. 
The slicer crashes with this dump  [crash.log](https://github.com/user-attachments/files/18051780/crash.log)

QIDITECH, 2024-12-10:
Thanks for your feedback, we will fix this issue in the next release.

wjyLearn, 2024-12-17:
@mssbg On MacOS x86 systems, when sending files, the program crashes when the file name is between 6 and 9 characters, which we have not yet found the reason. You can try to limit the file name to less than 6 characters or more than 9 characters before sending.