QIDITECH/QIDISlicer #112: Qidi Slicer 1.2.1 Send to printer option inconsistant

jonah0282, 2024-12-19:
Send to printer option disappears each time I close and re open Qidi slicer. I have have to remove printer from fluidd and re add for it to work again.

QIDITECH, 2024-12-20:
Sorry, we are unable to reproduce your question, please provide a video or picture to tell us what happened.

jonah0282, 2024-12-20:
Morning, I just done a quick video.
It shows the "send to printer" option, then I close down the slicer and
re-open it to show the "send to printer" option is gone, then I remove
my printer from fluidd then re add it to get the "send to printer option
back.
This happens on 4 different computers and seemed to start when I used
version 1.1.7
Thank you.

jonah0282, 2024-12-20:
https://photos.app.goo.gl/V633KGiVcAwfEPie8
Apols I'm unsure if a link to my drive for the video was attached

jonah0282, 2024-12-20:
I'm in work now trying to replicate the issue and so far I can't reproduce it. Maybe hold off for a bit and I'll do some more diagnosing on my end.

wjyLearn, 2024-12-20:
@jonah0282 We have seen the video you provided and it is indeed a bug. Before slicing you can select your Printer in "Printer" instead of using the default Pinter for slicing, after slicing "Send to printer" will appear.

As for this bug, we will fix it in the next release.

jonah0282, 2024-12-20:
Yeah that works, thank you.