QIDITECH/QIDISlicer #113: QIDI Slicer 1.2.1 Send to Printer option doesn't accept forward slash as directory separator

GusMcN, 2024-12-23:
Windows 11 Build22631.4602

I save my gcode files to a USB drive on my QIDI x-MAX 3. This drive appears on the printer as the subfolder sda1.
Since 1.2.,1 I can no longer save my files to this subdirectory as forward slash is not allowed. See attached screenshot.
[image]

QIDITECH, 2024-12-24:
Very sorry, this is our mistake, we will fix in the next version.

yperrey, 2024-12-28:
I confirm the same bug with QIDI Slicer 1.2.1 appimage with Ubuntu 24.04 LTS.

MiChAeLoKGB, 2025-01-25:
Still not fixed, this is very annoying guys, I separate every file I print into sub-folders, now I have to upload it to root and then move it manually through Fluidd interface.
The device definitely allows them, so it should be just a simple regex change, right (I really hope it's just that)?