QIDITECH/QIDISlicer #114: QIDISlicer 1.2.1 issue

joeycoates, 2024-12-24:
Just updated to 1.2.1 and QIDISlicer is now bricked.

QIDISlicer GUI initialization failed

Error parsing QIDISlicer config file, it is probably corrupted. Try to manually delete the file to recover from the error. Your user profiles will not be affected.

C;\\Users\willi\AppData\Roaming\QIDISlicer\QIDISlicer.ini

The configuration file contains null characters.

QIDITECH, 2024-12-25:
Sorry, we cannot reproduce the problem, please go to this path and send us the entire configuration folder.
C\Users\willi\AppData\Roaming\QIDISlicer\
Until we fix the problem, you can continue to use the software by deleting the configuration folder.

AndrewC110675, 2024-12-26:
I am having the same problem today and the link it gives doesnt exist.

AndrewC110675, 2024-12-26:
Where is the configuration folder because i for one cannot find it anywhere

QIDITECH, 2024-12-27:
AppDate is usually a hidden folder. Please enter the path to jump directly or display the hidden folder.

AndrewC110675, 2024-12-27:
I finally found it and deleted the file and all is working as it should.

joeycoates, 2024-12-27:
How did you find the config folder?

AndrewC110675, 2024-12-27:
Press the windows key plus R

AndrewC110675, 2024-12-27:
It will popup with %AppData% at bottom left,then go to qidi slicer and find config file and delete it.

AndrewC110675, 2024-12-27:
Let me know if you find it.

joeycoates, 2024-12-28:
Yeah, that did it, thanks! I was just about to reload 1.17.1 or just use OrcaSlicer (which works fine as well).

AndrewC110675, 2024-12-28:
Glad it worked out.

closed 2024-12-30 by QIDITECH