QIDITECH/QIDISlicer #115: Qidi Slicer Latest Version Crashing When Uploading Sliced File to Printer

found162, 2024-12-25:
Qidi Slicer Latest Version Crashing When Uploading Sliced File to Printer

I tried reinstalling same issue. I tried readding the printer. Same issue...

wjyLearn, 2024-12-26:
We apologize for the inconvenience to your experience. We can't analyze the specific cause of the software crash just by looking at the error screenshot you provided. We hope that you can provide the configuration information of your computer and the configuration file of the software. We will try to reproduce the problem. In the meantime, you can use QIDISlicer version 1.2.0 for the time being (if it works on your computer).

joseshiru, 2025-01-12:
Having the same issue in windows. It would be nice if a bit of QA is done before releasing new versions.

Jeykay, 2025-01-14:
Same here on Linux Mint 22 Wilma. Upload works but than it crashes. The same when i click on devices and than select my device (X-Max 3). First i see the Klippy-Dash with the message "Klippy not connected" and than, some seconds later, Qidi-Slicer crashes. Sorry for my bad english :-/

wjyLearn, 2025-01-17:
@Jeykay On some Linux systems, there is an issue where the Fluidd page is opened and the software crashes. This problem is caused by a bug in WebKit in some Linux versions and cannot be solved for the time being. You can only temporarily open the Fluidd page using the browser and upload GCode using the browser, or use the USB flash drive to save GCode. Sorry for the impact on your experience. We are actively looking for ways to solve this problem.

wjyLearn, 2025-01-17:
@joseshiru We have tested the functions of the software before the release of the version, but we may not be able to detect bugs in some functions due to the differences between different users' systems or configurations. If you encounter problems, you can provide corresponding system information and software configuration files or log files. This information can help us to reproduce the problem and optimize it.

Jeykay, 2025-01-18:
@wjyLearn OK, thx so far...