QIDITECH/QIDISlicer #117: Uploading gcode from Qidi Slicer 100X slower than it used to be XMax-3

trancified, 2025-01-13:
Not sure what happened, but on latest FW and qidi slicer versions, and when I used to be able to upload massive gcode files (over 20MB) in seconds, now they take multi minutes, often up to 10 min. This is from latest Win11 24H2 builds. Can't quite pinpoint when it started, but around end of Nov. Any ideas?

QIDITECH, 2025-01-14:
This should be a problem with the wifi card, please try switching to a wired network and see if it improves.

QIDITECH, 2025-01-15:
I am referring to the wifi card on the printer, switching to a wired network is the easiest test method to determine if the problem is in the wifi card. Or you can try transferring the wifi card from the motherboard to the usb port on the top of the printer.
If there is still no effect, please contact after-sales service.

mwqidiprintr, 2025-04-23:
I have recently started having the issue after months of no issue.  a few seconds between power cycle off/on and it seems fixed until the next print is done.   I am working on getting a wire to it to try again but months of no issue and all other wifi functions seem to work (deleting prints, settings, warming up print nozzle, camera images etc... ) I'm thinking I overloaded the local memory because I haven't deleted prints in a long time... Maybe it has a problem writing the files until the power cycle resets whatever that issue is and then it is fine.  but power cycle fixes in the meantime.  (annoyance)