QIDITECH/QIDISlicer #119: Clicking on QidiSlicer MacOS 1.2.1 doesn’t do anything

Kisitaqn, 2025-01-16:
I’ve clicked many times trying to get the new 1.2.1 for Mac OS. But it doesn’t do anything. I’m logged in, but still nothing.

wjyLearn, 2025-01-17:
Are you referring to the inability to download Mac OS 1.2.1 files from Release?

Kisitaqn, 2025-01-20:
Says I’m running QIDIslicer 1.1.7

Also I’m in QIDI slicer configuration/check for applications updates. That is where my issue is. I click multiple times. No joy

QIDITECH, 2025-01-21:
You can download the latest version of the software through this link.
https://github.com/QIDITECH/QIDISlicer/releases

Kisitaqn, 2025-01-21:
Thank you for your help.

Kisitaqn, 2025-01-21:
**New issue.**

“QIDISlicer” can’t be opened because Apple cannot check it for malicious software. This software needs to be updated. Contact the developer for more information.

QIDITECH, 2025-01-23:
Please turn on System Preferences - Security & Privacy - General  and selection still open.