QIDITECH/QIDISlicer #123: Qidi Slicer 1.2.1 cannot be launched on Ubuntu : core-dumped : libsoup-ERROR

doom-fr, 2025-01-26:
I got this error when launching Qidi Slicer 1.2.1

```
(process:227113): libsoup-ERROR **: 18:49:16.799: libsoup2 symbols detected. Using libsoup2 and libsoup3 in the same process is not supported.
(core dumped)
```

Is there a workaround ?

doom-fr, 2025-01-26:
Qidi Slicer seems to be for Ubuntu 20.04 which is more than 4 years old.

My Ubuntu :
```
§ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 24.04.1 LTS
Release:	24.04
Codename:	noble
```

QIDITECH, 2025-01-27:
Please try another appimage.
https://github.com/QIDITECH/QIDISlicer/releases
[image]

doom-fr, 2025-01-27:
Many thanks for the advice. Now, it works.

Qidi Slicer 1.1.7 update to 20.04 image even if Ubuntu is in 24.04 version.
Maybe add an Ubuntu version checking in the update process to select the good image ?
Or propose the different choices ?