QIDITECH/QIDISlicer #127: Spacemouse control customisations not applying unless prusa slicer started.

zyeborm, 2025-02-25:
This is a fairly minor issue, just a niggling irritation more than a show stopper.
I have a spacemouse ( https://3dconnexion.com/au/product/spacemouse-compact/ ) 
It works in quidi but the controls are inverted, I think it is using the default control scheme (mouse moves object rather than moving camera) I prefer the moves camera control scheme.
The spacemouse software picks up that quidi slicer is running so it should be applying the correct control scheme but for some reason it's still inverted.

The thing that makes me think this is a slicer not a spacemouse issue is. If I start prusa slicer even while quidi is running then the controls are fixed. 
This "fix" persists even after stopping and restarting quidi slicer without prusa running. Starting prusa slicer once per boot seems to fix it. 

I'm fairly sure there are specific hooks for spacemouse support in the base prusa slicer and I'm guessing there's some box not being ticked in the quidi version that applies the users control customization or something like that. My diagnosis is 100% guesswork though.

If there's anything I could do to help identify this issue, logs etc please let me know. Hopefully it'll be a simple fix.

(The main reason I got the spacemouse was to unify the control scheme across all my applications, fusion, slicers, kicad etc do I hold shift, middle mouse or right click to rotate the view in this one again?)

wjyLearn, 2025-02-26:
It would be great if you could provide a specific picture or video to illustrate the problem. Hopefully, there's a comparison between qidislicer and prusaslicer in the video that will shed some light on the problem. Or you can compare the Settings in **Preferences** between the two slicers.

zyeborm, 2025-03-06:
Made a quick video, sorry busy day so it's not super clear.
https://youtu.be/KvVec7YCTzo

I start by pushing the mouse forward and as you can see it rotates the object backwards.
I then start prusa slicer, you can see the mouse software pick up that it is running and apply the settings (which are mostly reversals)
I then close prusa and push the mouse forward again and the rotation is now correct (push forward go forward)

I tried reversing the inputs in the spacemouse software but it makes no difference to the rotation in the qidi software. Reversing the inputs in prusa does actually take effect in both qidi and prusa slicer. 

It seems like qidi is using the prusa slicer control settings even though the spacemouse software is detecting the applications separately and presenting separate configurations, starting prusa must load the config into the spacemouse software enough for qidi to use. I suspect qidi is identifying itself as prusa in some portion of how it interfaces with the spacemouse driver and things are getting confused as the application itself is being detected as qidi so it doesn't know which configuration to apply.