QIDITECH/QIDISlicer #128: Version 1.2.2 segfault without related Error message

maribox, 2025-02-27:
Both the Ubuntu20 and Ubuntu 24 AppImage (what even is the difference? Isn't the whole point of AppImage to only have one?) produce a segfault when trying to upload gcode to the printer. 

I do not have this issue when running 1.2.1.

I run Fedora KDE Spin, Kernel 6.12.9-200.fc41.x86_64

```
❯ ./QIDISlicer_1.2.2_Linux_Ubuntu24.AppImage
[2025-02-27 01:42:02.632438] [0x00007fba0ab09a80] [error]   UserAccount: Failed to read token - no datafile found.

** (qidi-slicer:11436): CRITICAL **: 01:42:02.948: Cannot register URI scheme wxfs more than once

** (qidi-slicer:11436): CRITICAL **: 01:42:02.948: Cannot register URI scheme memory more than once
[2025-02-27 01:42:02.978432] [0x00007fba0ab09a80] [error]   Unable to load bed model: /tmp/.mount_QIDISlDeOGjp/usr/bin/bin/../resources/profiles/QIDITechnology/X-Plus 4_bed.stl
[2025-02-27 01:42:03.096040] [0x00007fba0ab09a80] [error]   Unable to load bed model: /tmp/.mount_QIDISlDeOGjp/usr/bin/bin/../resources/profiles/QIDITechnology/X-Plus 4_bed.stl
[2025-02-27 01:42:04.855451] [0x00007fb9eeffd6c0] [error]   Failed to unzip file vendor_indices/QIDITechnology.ini. Directories are not supported. Skipping file.
[2025-02-27 01:42:04.856405] [0x00007fb9eeffd6c0] [error]   Failed to unzip file vendor_indices/vendor_indices/QIDITechnology.ini. Directories are not supported. Skipping file.
[2025-02-27 01:42:12.081352] [0x00007fba0ab09a80] [error]   Unable to load bed model: /tmp/.mount_QIDISlDeOGjp/usr/bin/bin/../resources/profiles/QIDITechnology/X-Plus 4_bed.stl
[2025-02-27 01:42:15.183446] [0x00007fb9527f56c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.183813] [0x00007fb94edff6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.184712] [0x00007fb952ff76c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.185669] [0x00007fb94edff6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.186519] [0x00007fb952ff76c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.187187] [0x00007fb94edff6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.188368] [0x00007fb94fbeb6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.189728] [0x00007fb950bee6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.190579] [0x00007fb94edff6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.191122] [0x00007fb9523f46c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.192135] [0x00007fb94fbeb6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.192634] [0x00007fb9537f96c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.193190] [0x00007fb953bfa6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.193937] [0x00007fb94edff6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.194656] [0x00007fb9543fc6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.195481] [0x00007fb9543fc6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.197024] [0x00007fb9547fd6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.199902] [0x00007fb9543fc6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.202094] [0x00007fb94edff6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.204334] [0x00007fb9513f06c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.207170] [0x00007fb9543fc6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.209212] [0x00007fb94edff6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.210803] [0x00007fb94edff6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.212829] [0x00007fb94edff6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.213986] [0x00007fb94edff6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.215798] [0x00007fb9517f16c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.217181] [0x00007fb94edff6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.218995] [0x00007fb950fef6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.220385] [0x00007fb94edff6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.222279] [0x00007fb9517f16c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.224082] [0x00007fb94edff6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.226599] [0x00007fb951bf26c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.228017] [0x00007fb94edff6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.229186] [0x00007fb954bfe6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.230644] [0x00007fb951ff36c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.232073] [0x00007fb954bfe6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.233707] [0x00007fb94edff6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.235341] [0x00007fb9513f06c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.237911] [0x00007fb9517f16c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.244318] [0x00007fb94edff6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.245459] [0x00007fb949dfe6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.246618] [0x00007fb951bf26c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.248307] [0x00007fb949dfe6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.249336] [0x00007fb949dfe6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.250360] [0x00007fb94edff6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.252506] [0x00007fb9537f96c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.253835] [0x00007fb94edff6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.255780] [0x00007fb953ffb6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.257125] [0x00007fb94edff6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.259132] [0x00007fb9523f46c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.260611] [0x00007fb94edff6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:42:15.261872] [0x00007fb97dfff6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[... 364 lines cut ...]
[... cut ...]
[2025-02-27 01:45:19.285578] [0x00007f3ba51fc6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:45:19.285906] [0x00007f3ba83ff6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:45:19.286272] [0x00007f3ba83ff6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[2025-02-27 01:45:19.286477] [0x00007f3ba83ff6c0] [error]   GCodeProcessor encountered an invalid toolchange (TIMELAPSE_TAKE_FRAME).
[1]    12487 segmentation fault (core dumped)  ./QIDISlicer_1.2.2_Linux_Ubuntu20.AppImage
```

wjyLearn, 2025-02-27:
The main difference between Ubuntu 20 and Ubuntu 24 lies in the version of the WebView dependency. Currently, there is a bug in higher versions of Linux where sending GCode tasks causes the software to crash. We are working hard to resolve this issue. You can try adding a port number: **7125**, when adding the printer, and then attempt to send the GCode (this has been effective in Ubuntu 24, but it is unclear for other versions).

jamcinnes2, 2025-03-02:
I think i'm seeing this as well. v1.2.2 appimage. I'm getting a segfault when going to the "Device" tab to connect to moonraker. This is in dmesg:

slic3r_main[23443]: segfault at 0 ip 00007f48d6921451 sp 00007ffc87287500 error 4 in libwebkit2gtk-4.0.so.37.71.8[b21451,7f48d6441000+3207000]

lieb, 2025-04-11:
Same here with v1.2.2 (also 1.2.1) running the AppImage on Fedora 41 x86_64. The printer is a QIDI X-Plus3 with v4.2.15 firmware. Same issue with the earlier version firmware.
When I attempt to connect I get what looks like a std device/klipper page but with an error "klippy not connected". It then hangs with nothing either in Slicer or the device page responding to mouse clicks. It then crashes. Since it hangs and I have to have a connected printer in order to "-" (delete) it, I have to edit/delete the .ini file...

I added socket 7125 to the URI per above and now I am talking to Moonraker which reports everything OK including "Klipper reports printer is ready". I click "-" to delete the printer and that works. Not clear this does much for me since the page does not have anything other than status.

wjyLearn, 2025-04-11:
@lieb  Perhaps you can use the following command to start:
~~~
WEBKIT_DISABLE_COMPOSITING_MODE=0 WEBKIT_DISABLE_DMABUF_RENDERER=0 ./QIDISlicer_1.2.3_Linux_Ubuntu24.AppImage
~~~
Use the above command to start, you can normally use the Fluidd page (do not need to add 7125 port number, valid on Ubuntu system)

lieb, 2025-04-11:
I saw this in the relnotes and added it to the command line. Things work now. tnx. Don't forget to note in the relnotes when you get a webkit version that works...