QIDITECH/QIDISlicer #130: QIDISlicer Has Encountered An Error (will not slice because of the update)

cbrown50, 2025-03-15:
Ran the Windows 64 update .exe, and now it fails to slice, everytime.   Shows an "Invalid Data" on the Export G-code button, and provides the below error message.

G-code export to C:\........\Temp\.12988_0.gcode failed due to invalid custom G-code sections:
start_gcode
Parsing error at line 1: Variable does not exist
PRINT_START BED=[first_layer_bed_temperature] HOTEND=[first_layer_temperature] CHAMBER=[volume_temperature]
                                                                                        ^
Please inspect the file C:\...........\Temp\.12988_0.gcode.tmp for error messages enclosed between
        !!!!! Failed to process the custom G-code template ...
and
        !!!!! End of an error report for the custom G-code template ...
for all macro processing errors.

wjyLearn, 2025-03-17:
The corresponding parameter variables have changed in the new version, which may make the old configuration unusable. You can start QIDISlicer after deleting the configuration file, and then add new configuration information.

Raxasus, 2025-04-17:
I got the same issue, been a while since I printed through so I got no idea what or where the old and new configuration file should be like for q1 pro

wjyLearn, 2025-04-18:
Perhaps the configuration of your Q1 Pro has been edited by yourself. Due to version updates, some parameters have changed. You will need to delete the old custom configuration and make new customizations based on the updated configuration.  

You can open the configuration folder by going to **Help -> Show Configuration Folder** in the software, then delete the old custom configuration files in the **printer** folder. (Make sure to close the software before deleting to avoid conflicts.)

[image]

[image]